leoshusar/4heat-esphome #2: Can't use it in an ESP32

jorgezazo, 2023-11-09:
If I try to install it into an ESP32, Process stop with this error trace:

________________

Compiling .pioenvs/hidroestufa-esp32/src/esphome/components/logger/logger.cpp.o
In file included from src/esphome/components/fourheat/fourheat.cpp:3:
src/esphome/components/fourheat/fourheat.h:4:10: fatal error: esphome/components/binary_sensor/binary_sensor.h: No such file or directory
#include “esphome/components/binary_sensor/binary_sensor.h”
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/hidroestufa-esp32/src/esphome/components/fourheat/fourheat.cpp.o] Error 1
In file included from src/esphome/components/fourheat/sensor/fourheat_sensor.h:4,
from src/esphome/components/fourheat/sensor/fourheat_sensor.cpp:3:
src/esphome/components/fourheat/fourheat.h:4:10: fatal error: esphome/components/binary_sensor/binary_sensor.h: No such file or directory
#include “esphome/components/binary_sensor/binary_sensor.h”
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/hidroestufa-esp32/src/esphome/components/fourheat/sensor/fourheat_sensor.cpp.o] Error 1
In file included from src/esphome/components/fourheat/helpers.h:3,
from src/esphome/components/fourheat/helpers.cpp:1:
src/esphome/components/fourheat/common.h:8:37: error: ‘optional’ is not a member of ‘std’
using parser_t = std::function<std::optional(const std::vector<uint8_t> &)>;
^~~~~~~~
src/esphome/components/fourheat/common.h:8:37: note: ‘std::optional’ is only available from C++17 onwards
src/esphome/components/fourheat/common.h:8:37: error: ‘optional’ is not a member of ‘std’
src/esphome/components/fourheat/common.h:8:37: note: ‘std::optional’ is only available from C++17 onwards
src/esphome/components/fourheat/common.h:8:47: error: template argument 1 is invalid
using parser_t = std::function<std::optional(const std::vector<uint8_t> &)>;
^
In file included from src/esphome/components/fourheat/helpers.cpp:1:
src/esphome/components/fourheat/helpers.h:18:86: error: ‘optional’ in namespace ‘std’ does not name a template type
const std::string get_query_datapoint_id(const std::string &datapoint_id, const std::optionalstd::string &query_datapoint_id);
^~~~~~~~
src/esphome/components/fourheat/helpers.h:18:81: note: ‘std::optional’ is only available from C++17 onwards
const std::string get_query_datapoint_id(const std::string &datapoint_id, const std::optionalstd::string &query_datapoint_id);
^~~
src/esphome/components/fourheat/helpers.h:18:94: error: expected ‘,’ or ‘…’ before ‘<’ token
const std::string get_query_datapoint_id(const std::string &datapoint_id, const std::optionalstd::string &query_datapoint_id);
^
In file included from src/esphome/components/fourheat/helpers.cpp:1:
src/esphome/components/fourheat/helpers.h:21:6: error: ‘optional’ in namespace ‘std’ does not name a template type
std::optional parse(const std::optional<parser_t> &parser, const std::vector<uint8_t> &data);
^~~~~~~~
src/esphome/components/fourheat/helpers.h:21:1: note: ‘std::optional’ is only available from C++17 onwards
std::optional parse(const std::optional<parser_t> &parser, const std::vector<uint8_t> &data);
^~~
src/esphome/components/fourheat/helpers.cpp:37:86: error: ‘optional’ in namespace ‘std’ does not name a template type
const std::string get_query_datapoint_id(const std::string &datapoint_id, const std::optionalstd::string &query_datapoint_id) {
^~~~~~~~
src/esphome/components/fourheat/helpers.cpp:37:81: note: ‘std::optional’ is only available from C++17 onwards
const std::string get_query_datapoint_id(const std::string &datapoint_id, const std::optionalstd::string &query_datapoint_id) {
^~~
src/esphome/components/fourheat/helpers.cpp:37:94: error: expected ‘,’ or ‘…’ before ‘<’ token
const std::string get_query_datapoint_id(const std::string &datapoint_id, const std::optionalstd::string &query_datapoint_id) {
^
src/esphome/components/fourheat/helpers.cpp: In function ‘const string esphome::fourheat::get_query_datapoint_id(const string&, int)’:
src/esphome/components/fourheat/helpers.cpp:38:7: error: ‘query_datapoint_id’ was not declared in this scope
if (query_datapoint_id.has_value()) {
^~~~~~~~~~~~~~~~~~
src/esphome/components/fourheat/helpers.cpp:38:7: note: suggested alternative: ‘get_query_datapoint_id’
if (query_datapoint_id.has_value()) {
^~~~~~~~~~~~~~~~~~
get_query_datapoint_id
src/esphome/components/fourheat/helpers.cpp: At global scope:
src/esphome/components/fourheat/helpers.cpp:49:6: error: ‘optional’ in namespace ‘std’ does not name a template type
std::optional parse(const std::optional<parser_t> &parser, const std::vector<uint8_t> &data) {
^~~~~~~~
src/esphome/components/fourheat/helpers.cpp:49:1: note: ‘std::optional’ is only available from C++17 onwards
std::optional parse(const std::optional<parser_t> &parser, const std::vector<uint8_t> &data) {
^~~
src/esphome/components/fourheat/helpers.cpp:64:6: error: ‘optional’ in namespace ‘std’ does not name a template type
std::optional parse(const std::optional<parser_t> &parser, const std::vector<uint8_t> &data) {
^~~~~~~~
src/esphome/components/fourheat/helpers.cpp:64:1: note: ‘std::optional’ is only available from C++17 onwards
std::optional parse(const std::optional<parser_t> &parser, const std::vector<uint8_t> &data) {
^~~
*** [.pioenvs/hidroestufa-esp32/src/esphome/components/fourheat/helpers.cpp.o] Error 1
========================= [FAILED] Took 14.05 seconds =========================

leoshusar, 2023-11-09:
Did you try to "clean build files" in ESPHome? I'm seeing similar issues when I add a new component to yaml I haven't had there before, ESPHome for some reason doesn't copy needed files. Cleaning fixes it for me.

jorgezazo, 2023-11-09:
I'll Give it a try and tell how it goes. Thank you

PR #4 (2023-11-10): Refactor offline sensor into binary sensor type
Moves offline sensor from `fourheat` component into `binary_sensor` type.

Before:
```yaml
fourheat:
  offline_sensor:
    name: Stove offline
```

After:
```yaml
fourheat:

binary_sensor:
  - platform: fourheat
    name: Stove offline
    type: module_offline
```

The reason is, ESPHome only includes `binary_sensor.h` if the top level `binary_sensor` exists, so the compilation fails, if you only use offline sensor and no other binary sensors.

Fixes #2
closed 2023-11-10 by leoshusar