leoshusar/4heat-esphome #5: After power off, can´t read stove's data again

jorgezazo, 2023-11-22:
Hi!

After many days working properly, I observed that after an energy cut off, the ESP can't read data from stove until the RS232 wire is disconnected and connected again. After that, it will work until next energy  cut off... Any clue on why this could be hapening?

BR

leoshusar, 2023-11-24:
Hi, do you mean cutting off power to ESP, stove or both?

jorgezazo, 2023-11-25:
To ESP. And another thing that has happened is, if I send a wrong command to the stove and it set an error 15, communication interrupts for ever. I mean, I can reboot the stove cutting power off and on, but communication is interrupted. I have to connect the 4Heat module and then it restablish the communication through RS232 and ESP works again :-?

leoshusar, 2023-11-25:
I had similar issues where the stove would stop responding when I sent a wrong command, but it fixed itself when I just rebooted it. I assumed it's just their bad command handling like it would lock itself for some reason, because the ESP should be sending right commands afterwards. And since the RS232 is idle high, maybe they are missing timeouts and pulling it low (ie. disconnecting RS232) fixes it? I don't really know :/

You are saying when you get error 15, it won't respond to ESP even after reboot until you connect their WiFi module? Huh, that's weird. The only idea I have now is to try to send those C E G commands, maybe it works as some sort of initialization in your board?

You can add these back and when it happens again, try to press them in order.

```yaml
button:
  - name: Button C
    platform: fourheat
    datapoint: C10000
    press_data: '000000000000'
  - name: Button E
    platform: fourheat
    datapoint: E10001
    press_data: '000000000000'
  - name: Button G
    platform: fourheat
    datapoint: G10000
    press_data: '000000000000'
```

Or if you still have the sniffer, try to log what the 4Heat module sends right after you connect it, if there is not something else.

But I know 100% when mine stopped responding due to wrong command, it didn't respond even to these, I had to reboot and then it worked fine. Maybe it's some control board difference.

jorgezazo, 2023-11-30:
Hi,

Sorry for the late answer but work rules. So the buttons doesn't work like a reset in case of lost communication. The logger... is unmmounted, have to mount it again and tray to look for the connection messages. It has happen twice more after first comment.

Thank you

jorgezazo, 2023-12-05:
Hi again.

This is the initial sequence between stove and 4Heat module. The communication with ESP module was not stuck yet... I'll catch it in one or two days.

```
[09:53:04][D][uart_debug:158]: <<< "\bC10000000000000000\r\bE10001000000000000\r\bG10000000000000000\r"
[09:53:04][D][uart_debug:158]: <<< "\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r"
[09:53:06][D][uart_debug:158]: <<< "\bI30001000000000000\r\bI30002000000000000\r\bI30005000000000000\r\bI30006000000000000\r\bA20816000000000000\r\bI30012000000000000\r\bI30017000000000000\r\bA20801000000000000\r\bI30020000000000000\r\bI30026000000000000\r\bI30038000000000000\r\bA20180000000000000\r\bA20211000000000000\r\bA20301000000000000\r\bA20211000000000000\r\bA20005000000000000\r\bA20006000000000000\r"
[09:53:06][D][uart_debug:158]: <<< "\bJ30001000000000011\r\bJ30002000000000000\r\bJ30005000000000015\r\bJ30006000000000000\r\bB20\x1816000000000006\r\bJ30012000000000000\r\bJ30013000000000022\r\bB20\x1801000000000001\r\bJ300200000000014\x196\r\bJ300260000000001\x190\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r\bB20301000000000001\r\bB20211000000000005\r\bB20005000000000060\r\bB200060000000000\x180\r"
```
The CEG sequence is launched and the SYEV command too... Do you see any difference with yours?

Thank you

leoshusar, 2023-12-05:
I don't see any difference, the CEG sequence is the same. Then what should matter is only what *you* send (like that CEG sequence), the response is irrelevant - unless the official module sends something more based on the response.

Those two other lines are normal datapoint query / responses, but they also contain the `\x1` in some random places... couldn't it be maybe faulty wiring or some interference? I haven't seen these anywhere in my logs. Then maybe sometimes the controller also receives some faulty data which locks it up? I don't know, just guessing.

jorgezazo, 2023-12-06:
Hi, it has get locked again and I unlock it with 4Heat Module. I attach another traces too

**Initialize which no blocking**
```
[09:53:04][D][uart_debug:158]: <<< "\bC10000000000000000\r\bE10001000000000000\r\bG10000000000000000\r"
[09:53:04][D][uart_debug:158]: <<< "\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r"
[09:53:06][D][uart_debug:158]: <<< "\bI30001000000000000\r\bI30002000000000000\r\bI30005000000000000\r\bI30006000000000000\r\bA20816000000000000\r\bI30012000000000000\r\bI30017000000000000\r\bA20801000000000000\r\bI30020000000000000\r\bI30026000000000000\r\bI30038000000000000\r\bA20180000000000000\r\bA20211000000000000\r\bA20301000000000000\r\bA20211000000000000\r\bA20005000000000000\r\bA20006000000000000\r"
[09:53:06][D][uart_debug:158]: <<< "\bJ30001000000000011\r\bJ30002000000000000\r\bJ30005000000000015\r\bJ30006000000000000\r\bB20\x1816000000000006\r\bJ30012000000000000\r\bJ30013000000000022\r\bB20\x1801000000000001\r\bJ300200000000014\x196\r\bJ300260000000001\x190\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r\bB20301000000000001\r\bB20211000000000005\r\bB20005000000000060\r\bB200060000000000\x180\r"
```

**Initialize after blocking** 

```
[05:35:43][D][uart_debug:158]: <<< "\x03\x00\bC10000000000000000\r\bE10001000000000000\r\bG10000000000000000\r"
[05:35:43][D][uart_debug:158]: <<< "\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r"
[05:35:46][D][uart_debug:158]: <<< "\bI30001000000000000\r\bI30002000000000000\r\bI30005000000000000\r\bI30006000000000000\r\bA20816000000000000\r\bI30012000000000000\r\bI30017000000000000\r\bA20801000000000000\r\bI30020000000000000\r\bI30026000000000000\r\bI30038000000000000\r\bA20180000000000000\r\bA20211000000000000\r\bA20301000000000000\r\bA20211000000000000\r\bA20005000000000000\r\bA20006000000000000\r"
[05:35:46][D][uart_debug:158]: <<< "\bJ30001000000000011\r\bJ30002000000000000\r\bJ30005000000000015\r\bJ30006000000000000\r\bB20\x1816000000000002\r\bJ30012000000000000\r\bJ30013000000000023\r\bB20\x1801000000000001\r\bJ30020000000001523\r\bJ300260000000001\x18\x18\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r\bB20301000000000001\r\bB20211000000000005\r\bB20005000000000060\r\bB200060000000000\x180\r"
```

**Initialize "complex"** 

```
[05:48:41][D][uart_debug:158]: <<< "\x00\x00\bC10000000000000000\r\bE10001000000000000\r\bG10000000000000000\r"
[05:48:41][D][uart_debug:158]: <<< "\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r"
[05:48:43][D][uart_debug:158]: <<< "\bI30001000000000000\r\bI30002000000000000\r\bI30005000000000000\r"
[05:48:43][D][uart_debug:158]: <<< "\bJ30001000000000011\r\bJ30002000000000000\r\bJ30005000000000014\r"
[05:48:53][D][uart_debug:158]: <<< "\bI30005000000000000\r\bI30006000000000000\r\bA20816000000000000\r\bI30012000000000000\r\bI30017000000000000\r"
[05:48:53][D][uart_debug:158]: <<< "\bJ30005000000000014\r\bJ30006000000000000\r\bB20\x1816000000000002\r\bJ30012000000000000\r\bJ30013000000000023\r"
[05:49:04][D][uart_debug:158]: <<< "\bI30017000000000000\r\bA20801000000000000\r\bI30020000000000000\r\bI30026000000000000\r\bI30038000000000000\r\bA20180000000000000\r\bA20211000000000000\r\bA20301000000000000\r\bA20211000000000000\r\bA20005000000000000\r\bA20006000000000000\r"
[05:49:04][D][uart_debug:158]: <<< "\bJ30013000000000023\r\bB20\x1801000000000001\r\bJ30020000000001523\r\bJ300260000000001\x18\x18\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r\bB20301000000000001\r\bB20211000000000005\r\bB20005000000000060\r\bB200060000000000\x180\r"
```
I attach too the ESP initialize and CEG command sending. I can't find any ```\x1``` in these ESP logs.

```
ESPHome Module request / response sequence sended every 30 seconds to check stove's state

[11:24:19][D][uart_debug:158]: <<< "\bJ30001000000000011\r\bJ30084000000000000\r\bJ30005000000000015\r\bJ30017000000000021\r\bJ30020000000001489\r\bJ30026000000000191\r\bJ30008000000000000\r\bJ30011000000000110\r\bJ30002000000000000\r\bB20180000000000065\r\bB20005000000000060\r\bB20006000000000080\r\bB20816000000000006\r"
[11:24:19][D][uart_debug:158]: <<< "\bI30001000000000000\r\bJ30001000000000011\r\bI30084000000000000\r\bJ30084000000000000\r\bI30005000000000000\r\bJ30005000000000015\r\bI30017000000000000\r\bJ30017000000000021\r\bI30020000000000000\r\bJ30020000000001489\r\bI30026000000000000\r\bJ30026000000000191\r\bI30008000000000000\r\bJ30008000000000000\r\bI30011000000000000\r\bJ30011000000000110\r\bI30002000000000000\r\bJ30002000000000000\r\bA20180000000000000\r\bB20180000000000065\r\bA20005000000000000\r\bB2000500000000

ESPHome Module CEG commands separated

[11:28:03][D][uart_debug:158]: <<< "\bD10000SYEVO0000544\r"
[11:28:03][D][uart_debug:158]: <<< "\bC10000000000000000\r\bD10000SYEVO0000544\r"
[11:28:06][D][uart_debug:158]: <<< "\bF10001000000000001\r"
[11:28:06][D][uart_debug:158]: <<< "\bE10001000000000000\r\bF10001000000000001\r"
[11:28:09][D][uart_debug:158]: <<< "\bH10202140820000041\r"
[11:28:09][D][uart_debug:158]: <<< "\bG10000000000000000\r\bH10202140820000041\r"

ESPHome Module CEG commands together

[11:36:48][D][uart_debug:158]: <<< "\bD10000SYEVO0000544\r\bF10001000000000001\r\bH10202140820000041\r"
[11:36:48][D][uart_debug:158]: <<< "\bC10000000000000000\r\bD10000SYEVO0000544\r\bE10001000000000000\r\bF10001000000000001\r\bG10000000000000000\r\bH10202140820000041\r"

ESPHome Module init commands

[... cut ...] J30002000000000000\r\bA20180000000000000\r\bB20180000000000065\r\bA20005000000000000\r\bB20005000000000060\r
```

The point here is the ```\x1``` I can't find it in any of ESP logs... so if the welding was welding I must find it in all logs... Strange things... The only difference in hardware is the RS232 5V line that is not connected to ESP but it is in 4Heat module. In fact, I didn't weld it in the 6C4P connectors at the beginning and 4Heat Module didn't work. The Module needs the 5V+ wire....

jorgezazo, 2023-12-06:
And these are the time goes on logs of the 4Heat Module... It ALWAYS send a CEG command before the stove's state request.

```
[05:50:26][D][uart_debug:158]: <<< "\bC10000000000000000\r\bE10001000000000000\r\bG10000000000000000\r"
[05:50:26][D][uart_debug:158]: <<< "\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r"
[05:50:28][D][uart_debug:158]: <<< "\bI30001000000000000\r\bI30002000000000000\r"
[05:50:28][D][uart_debug:158]: <<< "\bJ30001000000000011\r\bJ30002000000000000\r"
[05:50:38][D][uart_debug:158]: <<< "\bI30002000000000000\r\bI30005000000000000\r\bI30006000000000000\r\bA20816000000000000\r\bI30012000000000000\r\bI30017000000000000\r\bA20801000000000000\r\bI30020000000000000\r\bI30026000000000000\r\bI30038000000000000\r\bA20180000000000000\r\bA20211000000000000\r\bA20301000000000000\r\bA20211000000000000\r\bA20005000000000000\r\bA20006000000000000\r"
[05:50:39][D][uart_debug:158]: <<< "\bJ30002000000000000\r\bJ30005000000000014\r\bJ30006000000000000\r\bB20\x1816000000000002\r\bJ30012000000000000\r\bJ30013000000000023\r\bB20\x1801000000000001\r\bJ30020000000001523\r\bJ300260000000001\x18\x18\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r\bB20301000000000001\r\bB20211000000000005\r\bB20005000000000060\r\bB200060000000000\x180\r"
[05:52:01][D][uart_debug:158]: <<< "\bC10000000000000000\r\bE10001000000000000\r\bG10000000000000000\r"
[05:52:01][D][uart_debug:158]: <<< "\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r"
[05:52:03][D][uart_debug:158]: <<< "\bI30001000000000000\r\bI30002000000000000\r\bI30005000000000000\r\bI30006000000000000\r\bA20816000000000000\r\bI30012000000000000\r\bI30017000000000000\r\bA20801000000000000\r\bI30020000000000000\r\bI30026000000000000\r\bI30038000000000000\r\bA20180000000000000\r\bA20211000000000000\r"
[05:52:03][D][uart_debug:158]: <<< "\bJ30001000000000011\r\bJ30002000000000000\r\bJ30005000000000014\r\bJ30006000000000000\r\bB20\x1816000000000002\r\bJ30012000000000000\r\bJ30013000000000023\r\bB20\x1801000000000001\r\bJ30020000000001523\r\bJ300260000000001\x18\x18\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r"
[05:52:13][D][uart_debug:158]: <<< "\bA20211000000000000\r\bA20301000000000000\r\bA20211000000000000\r\bA20005000000000000\r\bA20006000000000000\r"
[05:52:13][D][uart_debug:158]: <<< "\bB20211000000000005\r\bB20301000000000001\r\bB20211000000000005\r\bB20005000000000060\r\bB200060000000000\x180\r"
[05:53:36][D][uart_debug:158]: <<< "\bC10000000000000000\r\bE10001000000000000\r\bG10000000000000000\r"
[05:53:36][D][uart_debug:158]: <<< "\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r"
[05:53:38][D][uart_debug:158]: <<< "\bI30001000000000000\r\bI30002000000000000\r\bI30005000000000000\r\bI30006000000000000\r\bA20816000000000000\r\bI30012000000000000\r\bI30017000000000000\r\bA20801000000000000\r\bI30020000000000000\r\bI30026000000000000\r\bI30038000000000000\r\bA20180000000000000\r\bA20211000000000000\r\bA20301000000000000\r\bA20211000000000000\r\bA20005000000000000\r\bA20006000000000000\r"
[05:53:38][D][uart_debug:158]: <<< "\bJ30001000000000011\r\bJ30002000000000000\r\bJ30005000000000014\r\bJ30006000000000000\r\bB20\x1816000000000002\r\bJ30012000000000000\r\bJ30013000000000023\r\bB20\x1801000000000001\r\bJ3002000000000152\x18\r\bJ300260000000001\x18\x18\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r\bB20301000000000001\r\bB20211000000000005\r\bB20005000000000060\r\bB200060000000000\x180\r"
[05:55:01][D][uart_debug:158]: <<< "\bC10000000000000000\r\bE10001000000000000\r\bG10000000000000000\r"
[05:55:01][D][uart_debug:158]: <<< "\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r"
[05:55:03][D][uart_debug:158]: <<< "\bI30001000000000000\r\bI30002000000000000\r\bI30005000000000000\r\bI30006000000000000\r\bA20816000000000000\r\bI30012000000000000\r\bI30017000000000000\r\bA20801000000000000\r\bI30020000000000000\r"
[05:55:03][D][uart_debug:158]: <<< "\bJ30001000000000011\r\bJ30002000000000000\r\bJ30005000000000014\r\bJ30006000000000000\r\bB20\x1816000000000002\r\bJ30012000000000000\r\bJ30013000000000023\r\bB20\x1801000000000001\r\bJ30020000000001530\r"
[05:55:13][D][uart_debug:158]: <<< "\bI30020000000000000\r\bI30026000000000000\r\bI30038000000000000\r\bA20180000000000000\r\bA20211000000000000\r\bA20301000000000000\r\bA20211000000000000\r\bA20005000000000000\r\bA20006000000000000\r"
[05:55:13][D][uart_debug:158]: <<< "\bJ30020000000001523\r\bJ300260000000001\x18\x18\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r\bB20301000000000001\r\bB20211000000000005\r\bB20005000000000060\r\bB200060000000000\x180\r"
[05:56:36][D][uart_debug:158]: <<< "\bC10000000000000000\r\bE10001000000000000\r\bG10000000000000000\r"
[05:56:36][D][uart_debug:158]: <<< "\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r"
[05:56:38][D][uart_debug:158]: <<< "\bI30001000000000000\r\bI30002000000000000\r\bI30005000000000000\r\bI30006000000000000\r\bA20816000000000000\r\bI30012000000000000\r\bI30017000000000000\r\bA20801000000000000\r\bI30020000000000000\r\bI30026000000000000\r\bI30038000000000000\r\bA20180000000000000\r\bA20211000000000000\r\bA20301000000000000\r\bA20211000000000000\r\bA20005000000000000\r\bA20006000000000000\r"
[... cut ...]
[06:07:03][D][uart_debug:158]: <<< "\bJ30001000000000011\r\bJ30002000000000000\r\bJ30005000000000014\r\bJ30006000000000000\r\bB20\x1816000000000002\r\bJ30012000000000000\r\bJ30013000000000026\r\bB20\x1801000000000001\r\bJ30020000000001523\r\bJ300260000000001\x18\x18\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r\bB20301000000000001\r\bB20211000000000005\r\bB20005000000000060\r\bB200060000000000\x180\r"
[06:08:05][D][uart_debug:158]: <<< "\xFE"
```

Would it be possible to implement a public method to request the CEG command before asking about the state?

Thank you very much

jorgezazo, 2023-12-07:
It works! Now I can disconnect the ESP module from current, reconnect it and it communicates with the stove through RS232 and start talking with no problem. I think we got it for this issue. If you want, let's leve it alone a couple of days and if all works OK, I'll let you know and you can "polish" the code and meka a new version.

Thank you so, so , so much for your work....

leoshusar, 2023-12-07:
Great! Sure, I'll wait.

It's still weird it doesn't work when you send the CEG manually using buttons, but if this fixes it, I won't complain.

jorgezazo, 2023-12-09:
Hi again Leo. After some time (much longer than the previous times), the stove blocks again and not even turning off and on it unblocks. After attaching the logger and turning on the 4Heat module, I got this dirty initialize sequence:

```
[07:57:36][D][uart_debug:158]: <<< "\x00\bC10000000000000000\r\bE10001000000000000\r\bG10000000000000000\r"
[07:57:36][D][uart_debug:158]: <<< "\bD\x11\x10\x10\x10\x10QIERA\x10\x10\x10\x10\x15\x14\x14\x05\bB\x11\x10\x10\x10\x11\x10\x10\x10\x10\x10\x10\x10\x10\x10\x10\x10\x11\x05\bH\x11\x10\x12\x10\x12\x11\x14\x10\b\x12\x10\x10\x10\x10\x10\x14\x11\x05"
[07:57:38][D][uart_debug:158]: <<< "\bI30001000000000000\r\bI30002000000000000\r\bI30005000000000000\r\bI30006000000000000\r\bA20816000000000000\r\bI30012000000000000\r\bI30017000000000000\r\bA20801000000000000\r\bI30020000000000000\r\bI30026000000000000\r\bI30038000000000000\r\bA20180000000000000\r\bA20211000000000000\r\bA20301000000000000\r\bA20211000000000000\r\bA20005000000000000\r\bA20006000000000000\r"
[07:57:38][D][uart_debug:158]: <<< "\bJ30001000000000011\r\bJ30002000000000000\r\bJ3000500000000001\x18\r\bJ30006000000000000\r\bB20\x1816000000000002\r\bJ30012000000000000\r\bJ30013000000000025\r\bB20\x1801000000000001\r\bJ30020000000001502\r\bJ300260000000001\x183\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r\bB20301000000000001\r\bB202110000000\x100005\r\bB2000500000000006\x10\r\bB200060000\x1000000\x180\r"
```

The one which took my eye was thie second line and after clean it a bit this was the result:

```
"\bD10000QIERA0000544\bB10001000000000001\bH10202140\b20000041"
```
And instead of "SYEVC" we get "QIERA"... The unblock chain maybe????

jorgezazo, 2023-12-09:
Oh! One more thing. In your code, the last "\r" and the closing double quote are missed... at least in this trace:

```
[08:00:06][D][uart_debug:158]: <<< "\bD10000SYEVO0000544\r\bF10001000000000001\r\bH10202140820000041\r\bJ30001000000000011\r\bJ30084000000000000\r\bJ30005000000000018\r\bJ30017000000000025\r\bJ30020000000001500\r\bJ30026000000000192\r\bJ30008000000000000\r\bJ30011000000000110\r\bJ30002000000000000\r\bB20180000000000065\r\bB20005000000000060\r\bB20006000000000080\r\bB20816000000000002\r"
[08:00:06][D][uart_debug:158]: <<< "\bC10000000000000000\r\bD10000SYEVO0000544\r\bE10001000000000000\r\bF10001000000000001\r\bG10000000000000000\r\bH10202140820000041\r\bI30001000000000000\r\bJ30001000000000011\r\bI30084000000000000\r\bJ30084000000000000\r\bI30005000000000000\r\bJ30005000000000018\r\bI30017000000000000\r\bJ30017000000000025\r\bI30020000000000000\r\bJ30020000000001500\r\bI30026000000000000\r\bJ30026000000000192\r\bI30008000000000000\r\bJ30008000000000000\r\bI30011000000000000\r\bJ3001100000000
[08:00:06][D][uart_debug:158]: <<< 
```

Thank you very much

jorgezazo, 2023-12-10:
Hi again... It still get stuck  after som working time and I can't find any intialization sequence... I wonder if the 5V wire is the key for this. At logger building, I didn't connect those pins between the 2 6P4C connectors and the 4 Heat module didn't connect to the stove even if the power wire through mini USB connector was attached, but after that, I welded those pins and then the 4Heat module start working properly. Maybe is this the key? Can you imagine some way to read what happens with this 5V wire at phisicall connection occurs?

Thak you in advance.

leoshusar, 2023-12-11:
Hi, so many news here!

> I got this dirty initialize sequence
> `"\bD10000QIERA0000544\bB10001000000000001\bH10202140\b20000041"`

But it's the stove controller response, isn't it? The module sends just CEG again.

> In your code, the last "\r" and the closing double quote are missed 
Do you have `rx_buffer_size` set to some high number? There are more characters missing than just `\r"`.

> I wonder if the 5V wire is the key for this
Everything is possible. But I don't see how, because (at least on my stove) the board sends 5V *to* the WiFi module and not the other way around. I think it's used so the WiFi module knows when the stove is connected so it can send it's CEG identification sequence.

Good luck with your experiments!

jorgezazo, 2023-12-13:
I think I've got it....

I have converted a GPIO pin into a current 3.3V pin. That gives me the chance to reboot the MAX232 module cutting power off and on after a couple of seconds. I do every time that ESP Module boots. With that feature, it has lasted all night long without any stuck. Now I use the actual version without the pre_query_sequence config and let's see what happens.

I have reattached the hat to the wemos with a relay to put the stove in standby and a climate object to set the temperature from it. I'm planning to attach it to a second wemos module with a belkin wemo plug simulator and a themometer to turn on the stove from alexa and use the thermometer sensor to implement a separated thermostat to regulate the stove's power and make a whole smart system to control it. That's why I need an ESP32 to have more process power and a bigger flash to implement all ESPHome modules to communicate through HTTP.

If the actual ESPHome module remains unstuck tomorrow, I'll telll you.

jorgezazo, 2023-12-13:
Aaaand no, I don´t get anything but a headache.... No way of maintain a stable connection for a long time... I 'm starting thinking about some faulty component. I will change the MAX232 module and let's see what happens... again. Tomorrow will be a new day...

leoshusar, 2023-12-13:
Well my theory is still with those random `\x` bytes. I *think* when the controller receives some, it locks the communication (because they are missing timeouts) and when you disconnect the cable, it releases the line and the controller UART resets.
But I don't know. It's really strange.

Good luck with your smart system!

jorgezazo, 2023-12-14:
Hi again. I want to make something with the 5V current from the estove but due to my lack of electronics knowledge I don't know how to connect it to a safe place in my breadboard. maybe as you say a couple of messages ago, is a way to "initialize" something ¿Connect it to the 5V pin of the wemos? ¿connect it to any resistance and add a led to use it? I'm lost with this...

jorgezazo, 2023-12-16:
Ok... I got it finally... and is no code and no stove's problems. Not one but TWO faulty components. The wemos 3V3 and 5V pins are faulty and current was stable at all and the MAX3232 was overheated and the result was to be burned. I redo all the job, and start from the beginning. I took another Wemos and another MAX3232 module and welded them again and all works 
now like a charm. No MAX3232 overheated and RX/TX communication absolutely right. So, I sorry for all your wasted time.... I'll remove the pre_query_sequence and will tell you the result to remove the branch if you wish.

Thank you a lot for your time.

jorgezazo, 2023-12-27:
Hi again. Now I'm in holydays and got time to work on this. The communication after days of working OK, suddenly stops and the only way to restart it was using the 4HEAT module (as always) Some kind of reboot command must be sended, but again, the only thing logger captures is this:

```
[07:39:40][D][uart_debug:158]: <<< "\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r"
[07:39:42][D][uart_debug:158]: <<< "\bJ30001000000000000\r\bJ30002000000000000\r\bJ30005000000000010\r\bJ30006000000000000\r\bB20\x1816000000000002\r\bJ30012000000000000\r\bJ30013000000000020\r\bB20\x1801000000000001\r\bJ30020000000001354\r\bJ300260000000001\x186\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r\bB20301000000000001\r\bB20211000000000005\r\bB20005000000000060\r\bB200060000000000\x180\r"
[07:41:05][D][uart_debug:158]: <<< "\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r"
[07:41:07][D][uart_debug:158]: <<< "\bJ30001000000000000\r\bJ30002000000000000\r\bJ30005000000000010\r\bJ30006000000000000\r\bB20\x1816000000000002\r\bJ30012000000000000\r\bJ30013000000000020\r"
[07:41:17][D][uart_debug:158]: <<< "\bJ30013000000000020\r\bB20\x1801000000000001\r\bJ30020000000001353\r\bJ300260000000001\x183\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r\bB20301000000000001\r\bB20211000000000005\r\bB20005000000000060\r\bB200060000000000\x180\r"
[07:42:40][D][uart_debug:158]: <<< "\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r"
[07:42:42][D][uart_debug:158]: <<< "\bJ30001000000000000\r\bJ30002000000000000\r\bJ30005000000000010\r\bJ30006000000000000\r\bB20\x1816000000000002\r\bJ30012000000000000\r\bJ30013000000000020\r\bB20\x1801000000000001\r\bJ3002000000000135\x18\r\bJ300260000000001\x186\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r\bB20301000000000001\r\bB20211000000000005\r\bB20005000000000060\r\bB200060000000000\x180\r"
[07:44:05][D][uart_debug:158]: <<< "\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r"
[07:44:07][D][uart_debug:158]: <<< "\bJ30001000000000000\r\bJ30002000000000000\r\bJ30005000000000010\r\bJ30006000000000000\r\bB20\x1816000000000002\r\bJ30012000000000000\r\bJ30013000000000020\r\bB20\x1801000000000001\r\bJ30020000000001355\r\bJ300260000000001\x183\r\bJ3003\x18000000000000\r\bB201\x180000000000065\r\bB20211000000000005\r\bB20301000000000001\r\bB20211000000000005\r"
```
Those \x18 are always in the same place...

I added a "RS232" restart button with the initial sequence os SYEV commands.

```
  - id: send_restart_command
    name: Restart stove's communication
    platform: template
    on_press:
      - uart.write: '\bD10000SYEVC0000544\r\bF10001000000000001\r\bH10202140\x1820000041\r'
```

 I don't know if that is sended or received by 4HEAT module, but the logger doesn't log any other thing at the communication starts...

leoshusar, 2024-01-04:
I'm afraid I'm out of ideas here and I still think those \x shouldn't be there :D

Those DFH commands are response from the board, not something you should send from the ESP.
D10000: SYEVC0000544 is the product code
F10001: 000000000001 is the product version
H102021401820000041 I *think* this reads:
```
H1020 - don't know
     2 - 4HEAT FW version
      1401820000041 - Product revision
```

It matches mine when I check the 4Heat Android app:
D10000: SYEVO0000562
F10001: 000000000001
H1020: 4 030921000069
[image]

closed 2024-11-09 by leoshusar