leoshusar/4heat-esphome #6: Error at ESP32-C3 install

jorgezazo, 2023-12-11:
Hi again.

4Heat-ESPhome can´t be installed in ESP32-C3

This are the first traces of error:

In file included from src/esphome/components/fourheat/helpers.h:3,
                 from src/esphome/components/fourheat/helpers.cpp:1:
src/esphome/components/fourheat/common.h:8:37: error: 'optional' is not a member of 'std'
 using parser_t = std::function<std::optional<T>(const std::vector<uint8_t> &)>;
                                     ^~~~~~~~
src/esphome/components/fourheat/common.h:8:37: note: 'std::optional' is only available from C++17 onwards
src/esphome/components/fourheat/common.h:8:37: error: 'optional' is not a member of 'std'
src/esphome/components/fourheat/common.h:8:37: note: 'std::optional' is only available from C++17 onwards
src/esphome/components/fourheat/common.h:8:47: error: template argument 1 is invalid
 using parser_t = std::function<std::optional<T>(const std::vector<uint8_t> &)>;

leoshusar, 2023-12-11:
Did you try to clear build files?

jorgezazo, 2023-12-12:
Yes. But no luck with that...

leoshusar, 2023-12-12:
Can you post your config? Sensors and stuff are not needed, only the basics.

jorgezazo, 2023-12-12:
YEs of course

```
esphome:
  name: hidroestufa-2
  friendly_name: Hidroestufa_2

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: ""

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Hidroestufa-2 Fallback Hotspot"
    password: "a1b2c3d4"

#captive_portal:

web_server:
  port: 8008
  local: true
  
external_components:
  source: github://leoshusar/4heat-esphome@feat/pre-query-sequence
  #source: github://leoshusar/4heat-esphome

uart:
  tx_pin: 21
  rx_pin: 20
  baud_rate: 9600

fourheat:
  update_interval: 10s
#  pre_query_sequence:
#    - C10000
#    - E10001
#    - G10000

switch:
  - id: state_switch
    name: State
    platform: fourheat
    restore_mode: DISABLED
    datapoint: J30001
    on_datapoint: J30253
    off_datapoint: J30254
    on_data: '1'
    off_data: '1'
    parser: |-
      if (id(state) != 0) {
        return true;
      } else {
        return false;
      }

#  - platform: gpio
#    id: standby_relay
#    pin: 5
#    name: "Standby relé"

binary_sensor:
  - id: water_pump_state
    name: Water pump state
    platform: fourheat
    datapoint: J30084

  - id: stove_offline
    name: Stove offline
    platform: fourheat
    type: module_offline
  
sensor:

  - id: exhaust_temperature
    name: Exhaust temperature
    platform: fourheat
    datapoint: J30005
    device_class: temperature
    unit_of_measurement: °C

  - id: boiler_temperature
    name: Boiler temperature
    platform: fourheat
    datapoint: J30017
    device_class: temperature
    unit_of_measurement: °C

  - id: water_pressure
    name: Water pressure
    platform: fourheat
    datapoint: J30020
    device_class: pressure
    unit_of_measurement: mbar

  - id: combustion_fan
    name: Combustion Fan Speed
    platform: fourheat
    datapoint: J30008
    device_class: speed
    unit_of_measurement: rpm

  - id: combustion_power
    name: Combustion Power
    platform: fourheat
    datapoint: J30011
    device_class: power_factor
    unit_of_measurement: ""

#  - platform: homeassistant
#    id: temperature_sensor
#    name: "Sensor de temperatura"
#    entity_id: sensor.netatmo_zazos_manor_interior_temperature

text_sensor:
  - id: state
    name: State
    platform: fourheat
    datapoint: J30001
    options:
      0: "OFF"
      1: "Check Up"
      2: "Ignition"
      3: "Stabilization"
      4: "Ignition"
      5: "Run"
      6: "Modulation"
      7: "Extinguishing"
      8: "Safety"
      9: "Block"
      10: "Recover Ignition"
      11: "Standby"
      30: "Ignition"
      31: "Ignition"
      32: "Ignition"
      33: "Ignition"
      34: "Ignition"

  - id: error
    name: Error
    platform: fourheat
    datapoint: J30002
    options:
      0: "No"
      1: "Safety Thermostat HV1: signalled also in case of Stove OFF"
      2: "Safety PressureSwitch HV2: signalled with Combustion Fan ON"
      3: "Extinguishing for Exhausting Temperature lowering"
      4: "Extinguishing for water over Temperature"
      5: "Extinguishing for Exhausting over Temperature"
      6: "unknown"
      7: "Encoder Error: No Encoder Signal (in case of P25=1 or 2)"
      8: "Encoder Error: Combustion Fan regulation failed (in case of P25=1 or 2)"
      9: "Low pressure in to the Boiler"
      10: "High pressure in to the Boiler Error"
      11: "DAY and TIME not correct due to prolonged absence of Power Supply"
      12: "Failed Ignition"
      13: "Ignition"
      14: "Ignition"
      15: "Lack of Voltage Supply"
      16: "Ignition"
      17: "Ignition"
      18: "Lack of Voltage Supply"

number:
  - id: boiler_target_temperature
    name: Boiler target temperature
    platform: fourheat
    datapoint: B20180
    device_class: temperature
    unit_of_measurement: °C
    min_value: 50
    max_value: 90
  
  - id: set_boiler_minimum_temperature
    name: Boiler minimum temperature
    platform: fourheat
    datapoint: B20005
    device_class: temperature
    unit_of_measurement: °C
    min_value: 55
    max_value: 65

  - id: set_boiler_maximum_temperature
    name: Boiler maximum temperature
    platform: fourheat
    datapoint: B20006
    device_class: temperature
    unit_of_measurement: °C
    min_value: 75
    max_value: 80

button:
  - id: clear_error
    name: Clear error
    platform: fourheat
    datapoint: J30255

select:
  - id: power
    name: Power
    platform: fourheat
    datapoint: B20816
    options:
      1: "P1"
      2: "P2"
      3: "P3"
      4: "P4"
      5: "P5"
      6: "Auto"
 
#climate:
#  - platform: thermostat
#    name: "Thermostat Climate Controller"
#    sensor: temperature_sensor
#    min_heating_off_time: 300s
#    min_heating_run_time: 300s
#    min_idle_time: 30s
#    heat_action:
#      - switch.turn_on: standby_relay
#    idle_action:
#      - switch.turn_off: standby_relay
#    default_preset: Home
#    preset:
#      - name: Home
#        default_target_temperature_low: 21 °C
#      - name: Away
#        default_target_temperature_low: 16 °C
#      - name: Sleep
#        default_target_temperature_low: 18 °C

```

leoshusar, 2023-12-13:
So I don't know what C++ version the Arduino framework uses, since it says `'std::optional' is only available from C++17 onwards`. Or maybe it's a bug in ESPHome, because the ESP32-C3 was added fairly recently.

But I managed to compile using ESP-IDF:
```yaml
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    platform_version: https://github.com/platformio/platform-espressif32
    version: 5.1.1
```

Let me know if everything works fine with this.

leoshusar, 2024-01-04:
ESP should log something when booting if you connect it via USB or one of the UART buses. There should be something even if the flashed binary is faulty anyhow.

Mancio-home, 2024-03-10:
esphome:
  name: stufa
  friendly_name: stufa
  platformio_options:
    build_flags:
      - "-std=gnu++17"
    build_unflags:
      - "-std=gnu++11"
esp32:
  board: esp32dev
  framework:
    type: arduino
#esp32:
#  board: esp32dev
 # framework:
 #   type: arduino

wifi:
  ssid:
  password: 
  ap:
    ssid: "AP-Stufa"
    password:

# Enable logging
logger:
  level: DEBUG
  hardware_uart: UART1

# Enable Home Assistant API
api:
  encryption:
    key: 

ota:
  password:
  safe_mode: True

    # Enable fallback hotspot (captive portal) in case wifi connection fails

#captive_portal:

web_server:
  port: 8008
  local: true
    
external_components:  
  source: github://leoshusar/4heat-esphome

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 9600

fourheat:

switch:
  - id: state_switch
    name: Stato
    platform: fourheat
    restore_mode: DISABLED
    datapoint: J30001
    on_datapoint: J30253
    off_datapoint: J30254
    on_data: '1'
    off_data: '1'
    parser: |-
      if (id(state) != 0) {
        return true;
      } else {
        return false;
      }

binary_sensor:
  - id: water_pump_state
    name: Pompa Acqua
    platform: fourheat
    datapoint: J30084

  - id: stove_offline
    name: Stufa offline
    platform: fourheat
    type: module_offline
  
sensor:

  - id: exhaust_temperature
    name: Temperatura di scarico
    platform: fourheat
    datapoint: J30005
    device_class: temperature
    unit_of_measurement: °C

  - id: boiler_temperature
    name: Temperatura Boiler
    platform: fourheat
    datapoint: J30017
    device_class: temperature
    unit_of_measurement: °C

  - id: water_pressure
    name: Pressione Acqua
    platform: fourheat
    datapoint: J30020
    device_class: pressure
    unit_of_measurement: mbar

  - id: puffer_temperature
    name: Temperatura Puffer
    platform: fourheat
    datapoint: J30026
    device_class: temperature
    unit_of_measurement: °C

  - id: combustion_fan
    name: Velocità combustione
    platform: fourheat
    datapoint: J30008
    device_class: speed
    unit_of_measurement: rpm

  - id: combustion_power
    name: Potenza Combustione
    platform: fourheat
    datapoint: J30011
    device_class: power_factor
    unit_of_measurement: ""

text_sensor:
  - id: state
    name: Stato
    platform: fourheat
    datapoint: J30001
    options:
      0: "OFF"
      1: "Check Up"
      2: "Ignition"
      3: "Stabilization"
      4: "Ignition"
      5: "Run"
      6: "Modulation"
      7: "Extinguishing"
      8: "Safety"
      9: "Block"
      10: "Recover Ignition"
      11: "Standby"
      30: "Ignition"
      31: "Ignition"
      32: "Ignition"
      33: "Ignition"
      34: "Ignition"

  - id: error
    name: Errori
    platform: fourheat
    datapoint: J30002
    options:
      0: "No"
      1: "Safety Thermostat HV1: signalled also in case of Stove OFF"
      2: "Safety PressureSwitch HV2: signalled with Combustion Fan ON"
      3: "Extinguishing for Exhausting Temperature lowering"
      4: "Extinguishing for water over Temperature"
      5: "Extinguishing for Exhausting over Temperature"
      6: "unknown"
      7: "Encoder Error: No Encoder Signal (in case of P25=1 or 2)"
      8: "Encoder Error: Combustion Fan regulation failed (in case of P25=1 or 2)"
      9: "Low pressure in to the Boiler"
      10: "High pressure in to the Boiler Error"
      11: "DAY and TIME not correct due to prolonged absence of Power Supply"
      12: "Failed Ignition"
      13: "Ignition"
      14: "Ignition"
      15: "Lack of Voltage Supply"
      16: "Ignition"
      17: "Ignition"
      18: "Lack of Voltage Supply"

number:
  - id: boiler_target_temperature
    name: Temperatura targhet Boiler
    platform: fourheat
    datapoint: B20180
    device_class: temperature
    unit_of_measurement: °C
    min_value: 50
    max_value: 90
  
  - id: set_boiler_minimum_temperature
    name: Temperatura minima Boiler
    platform: fourheat
    datapoint: B20005
    device_class: temperature
    unit_of_measurement: °C
    min_value: 55
    max_value: 65

  - id: set_boiler_maximum_temperature
    name: Temperatura Massima Boiler
    platform: fourheat
    datapoint: B20006
    device_class: temperature
    unit_of_measurement: °C
    min_value: 75
    max_value: 80

button:
  - id: clear_error
    name: Reset Errori
    platform: fourheat
    datapoint: J30255

select:
  - id: power
    name: Potenza
    platform: fourheat
    datapoint: B20816
    options:
      1: "P1"
      2: "P2"
      3: "P3"
      4: "P4"
      5: "P5"
      6: "Auto"

Ciao, io ho questa configurazione caricata su esphome . non funziona, non aggiorna nulla.

leoshusar, 2024-03-12:
Hi @Mancio-home, anything in logs?

Mancio-home, 2024-03-13:
[logs_stufa_logs (1).txt](https://github.com/leoshusar/4heat-esphome/files/14594720/logs_stufa_logs.1.txt)

uso un ESP32 d1 mini.

ho sul mio home assistant l ' integrazione 4heat. il dispositivo si connette regolarmente ed è riconosciuto, ma non aggiorna sensori e non funziona ne pulsante reset errori e ne l'interruttore di accensione.

leoshusar, 2024-03-13:
Hmm, this seems to be some ESPHome issue, same errors are [here](https://github.com/esphome/issues/issues/5386) and [here](https://github.com/esphome/issues/issues/5384).

Can you try to get better logs via USB?

Mancio-home, 2024-03-14:
quindi potrebbe essere per questo motivo che l integrazio 4heat non funzioni? mi ha registrato solo 2 entità, lo switch on/off ed il pulsante del reset errori.
qualcuno ha una soluzione?

leoshusar, 2024-03-14:
It's hard to say, but maybe?

Do you have a different ESP32 which is not C3? Something fully supported by ESPHome? You could try with that, it should say if the issue is ESP32-C3 or something else.

Mancio-home, 2024-03-14:
no io ho un ESP32 D1 MINI.

tuisblad, 2024-04-05:
Same issues here.

ESP32 with RS232 from M5Stack [https://shop.m5stack.com/products/atom-rs232-kit?variant=34787913105572](url)
Stove is from Dielle, I use normally [https://github.com/zaubererty/homeassistant-4heat](url) but I am not happy with the slow response and the time-outs.
YAML and LOG files are attached [fourheat.zip](https://github.com/leoshusar/4heat-esphome/files/14885312/fourheat.zip)

leoshusar, 2024-04-05:
@tuisblad can you try with newer IDF version?
Something like:

```yaml
framework:
    type: esp-idf
    version: 5.1.1
```

tuisblad, 2024-04-13:
I am getting closer, the build flags give results:

esphome:
  name: fourheat
  friendly_name: dielle
  platformio_options:
    build_flags:
      - "-std=gnu++17"
    build_unflags:
      - "-std=gnu++11"
      
esp32:
  board: m5stack-core-esp32
  framework: 
    type: esp-idf

closed 2025-05-25 by leoshusar