leoshusar/4heat-esphome #7: ESP32 D1 MINI comunicazione fallita

Mancio-home, 2024-03-16:
ciao,
ho problemi con il funzionamento.
il mio dispositivo è un ESP32 D1 MINI collegato a un convertitore ttl232. (max 232).
[image]
 questo è il mio dispositivo, e anche se il GPIO3 è TX  ed il GPIO1 è il Rx , sono costretto ad invertirli per vedere qualcosa.
di seguito il mio Yaml:

esphome:
  name: stufa
  friendly_name: stufa
  platformio_options:
    build_flags:
      - "-std=gnu++17"
    build_unflags:
      - "-std=gnu++11"
esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: "xxxxxxx"
  password: "xxxxxxx"
  ap:
    ssid: "AP-Stufa"
    password: "xxxxxx"

# Enable logging
#logger:
  #level: DEBUG
  #hardware_uart: UART1

# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxx"

ota:
  password: "xx"
  safe_mode: True

    # Enable fallback hotspot (captive portal) in case wifi connection fails

#captive_portal:

web_server:
  port: 8008
  local: true
    
external_components:  
  source: github://leoshusar/4heat-esphome

uart:
  tx_pin: GPIO1
  rx_pin: GPIO3
  baud_rate: 9600

fourheat:

switch:
  - id: state_switch
    name: Stato
    platform: fourheat
    restore_mode: DISABLED
    datapoint: J30001
    on_datapoint: J30253
    off_datapoint: J30254
    on_data: '1'
    off_data: '1'
    parser: |-
      if (id(state_switch) != 0) {
        return true;
      } else {
        return false;
      };

text_sensor:
  - id: state
    name: Stato
    platform: fourheat
    datapoint: J30001
    options:
      0: "OFF"
      1: "Check Up"
      2: "Ignition"
      3: "Stabilization"
      4: "Ignition"
      5: "Run"
      6: "Modulation"
      7: "Extinguishing"
      8: "Safety"
      9: "Block"
      10: "Recover Ignition"
      11: "Standby"
      30: "Ignition"
      31: "Ignition"
      32: "Ignition"
      33: "Ignition"
      34: "Ignition"

di seguito ti allego l immagine dello sniff, a sinistra sono collegato direttamente all'USB del dispositivo e a destra con un convertitore sono collegato al TX della stufa.
[image]

come puoi vedere ESP invia il codice di richiesta dello stato, e la stufa risponde correttamente ( in questo momento è accesa).
Purtroppo lo stato non viene aggiornato mentre dovrebbe essere in "run":

[image]

ed inoltre se premo lo switch non succede nulla, nemmeno dallo sniff cambia il codice di richiesta.

l'installazione sul dispositivo non ha dato alcun problema.

grazie

Mancio-home, 2024-03-16:
Piccolo aggiornamento, ho provato a scrivere dal lato del ESP dei codici .J30254000000000001. e si è interrotta la comunicazione, poi agendo sullo switch è ripresa la comunicazione da parte dell'ESP con il codice di accensione ( la stua è già accesa).

[image]

lo stato contina a non aggiornarsi...

leoshusar, 2024-03-16:
Hi, you have your switch parser wrong.

The parser code takes your stove's response and then decides if it will make the switch on or off.
Your code currently does this:
- receive data
- check if the switch is on or off and return true or false (but there is no logic which actually does something with the switch)

And you need:
- receive `data`, it's an array of chars (`['J','3','0','0','0','1','0','0','0','0','0','0','0','0','0','0','X','X']`)
- check what the `data` says
- if it says it's on, return true, so the switch knows it should show it's on

Something like this, from my code:
```yaml
switch:
  - ...
    datapoint: J30001
    ...
    parser: |-
      auto size = data.size();
      auto last = data[size - 1];

      auto off =
        data[size - 2] == '0' &&
        (last == '0' || last == '7');

      return !off;
```

This code checks if the returned status is either `J30001000000000000` (off) or `J30001000000000007` (extinguishing) and tells the switch to turn "on" if it's not true (basically everything except 00 and 07 turns the switch on).

Mancio-home, 2024-03-21:
Ciao, solo ora ho potuto effettuare un Test.
ho trovato il convertitore TTL - RS232  guasto. Dopo la sostituzione ho implementato il tuo codice.
Ora si aggiornano alcuni stati e lo switch. i stati che mi interessano ci sono tutti.
vedo però dal mio home assistant che mi arriva di continuo un "problema" sullo stato online/offline.
Inoltre non funziona accensione e spegnimento.
Se provo a dare un comando ( reset errori / on off / potenza combustione) l ESP si blocca non aggiornando piu i stati.
ora proverò a eliminare tutto quello che non mi serve.
hai qualche consiglio sul blocco?

leoshusar, 2024-03-23:
Hi, great you are making some progress.

Anything in logs? Can you check with verbose log?

Mancio-home, 2024-03-28:
Ciao, dopo le modifiche ricevo correttamente gli stati ma si blocca il dispositivo appena invio un comando, anche l aggiornamento dei stati si bloccano.
di seguito il log dall ESP:

[image]

Attualmente la configurazione è questa :

esphome:
  name: stufa
  friendly_name: stufa
  platformio_options:
    build_flags:
      - "-std=gnu++17"
    build_unflags:
      - "-std=gnu++11"
esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  ssid: "*********"
  password: "*********"
  ap:
    ssid: "AP-Stufa"
    password: "********"

# Enable logging
logger:
  level: DEBUG
  hardware_uart: UART0
  

# Enable Home Assistant API
api:
  encryption:
    key: "*********"

ota:
  password: "********"
  safe_mode: True

    # Enable fallback hotspot (captive portal) in case wifi connection fails

#captive_portal:

web_server:
  port: 8008
  local: true
    
external_components:  
  source: github://leoshusar/4heat-esphome

uart:
  
    tx_pin: GPIO3
    rx_pin: GPIO1
    baud_rate: 9600

fourheat:

switch:
  - id: state_switch
    name: 2..Stato
    platform: fourheat
    restore_mode: DISABLED
    datapoint: J30001
    on_datapoint: J30253
    off_datapoint: J30254
    on_data: '1'
    off_data: '1' 
    parser: |-
      auto size = data.size();
      auto last = data[size - 1];

      auto off =
        data[size - 2] == '0' &&
        (last == '0' || last == '7');
      return !off;

text_sensor:
  - id: state
    name: 1..Stato
    platform: fourheat
    datapoint: J30001
    options:
      0: "OFF"
      1: "Check Up"
      2: "Accensione 1"
      3: "Stabilization"
      4: "Accensione 2"
      5: "Accesa"
      6: "Modulazione"
      7: "Spegnimento"
      8: "Safety"
      9: "Block"
      10: "Recupero Accensione"
      11: "Standby"
      30: "Ignition"
      31: "Ignition"
      32: "Ignition"
      33: "Ignition"
      34: "Ignition"

  - id: error
    name: 3..Errori
    platform: fourheat
    datapoint: J30002
    options:
      0: "Nessun Errore"
      1: "Safety Thermostat HV1: signalled also in case of Stove OFF"
      2: "Safety PressureSwitch HV2: signalled with Combustion Fan ON"
      3: "Spegnimento per Abbassamento Temperatura Fumi"
      4: "Spegnimento per acqua sovratemperatura"
      5: "Spegnimento per esaurimento per sovratemperatura"
      6: "unknown"
      7: "Spegimento per mancata Accensione" #errore 12 sulla stufa
      8: "Encoder Error: Combustion Fan regulation failed (in case of P25=1 or 2)"
      9: "Bassa pressione nella caldaia"
      10: "Alta pressione nella caldaia"
      11: "GIORNO e ORA non corretti a causa di prolungata assenza di alimentazione"
      12: "Accensione fallita"
      13: "Ignition"
      14: "Ignition"
      15: "Mancanza di alimentazione"
      16: "Ignition"
      17: "Ignition"
      18: "Lack of Voltage Supply"

button:
  - id: clear_error
    name: 4..Reset Errori
    platform: fourheat
    datapoint: J30255

select:

  - id: power
    name: 6..Seleziona Potenza
    platform: fourheat
    datapoint: B20816
    options:
      1: "P1"
      2: "P2"
      3: "P3"
      4: "P4"
      5: "P5"
      6: "Auto"

sensor:

  - id: combustion_power
    name: 5..Potenza Attuale
    platform: fourheat
    datapoint: J30011
    device_class: power_factor
    unit_of_measurement: ""

  - id: exhaust_temperature
    name: 8..Temperatura scarico
    platform: fourheat
    datapoint: J30005
    device_class: temperature
    unit_of_measurement: °C

  - id: boiler_temperature
    name: 7..Temperatura Termosifoni
    platform: fourheat
    datapoint: J30017
    device_class: temperature
    unit_of_measurement: °C

leoshusar, 2024-03-28:
Maybe your stove uses different commands, [here](https://github.com/leoshusar/4heat-esphome/issues/3#issuecomment-1807196760) was a similar issue.
Also you can try [these](https://github.com/neuralldev/homeassistant-4heat/blob/master/custom_components/4heat/const.py) commands.

Can you identify what board model do you have? You should be able to determine just by looking at connector positions https://www.tiemmeelettronica.com/en/categoria-prodotto/system-en/mother-board-en/

Mancio-home, 2024-03-30:
Ciao, la scheda della stufa è una SY250.
ho fatto ulteriori verifiche e questo è l'esito dei log:

[image]

a quanto pare alcuni codici non piacciono. non sono sicuro ma credo di averne estratti alcuni:

41.32.30.38.31.36.30.30.30.30.30.30.30.30.30.30.30.30.
A2081600000000000000

avvolte piu di una richiesta fallisce ma alla fine la stufa risponde con il dato, solamente dà fastidio.

Inoltre quando provo a accendere/spengere o variare la potenza, la stufa si spenge immediatamente, di seguito ti mostro cosa invia ESP:

[image]

Dopo l' invio dei comandi sono costretto a spengere e riaccendere.

questo è quello che succede quando riavvio l esp:

[image]

x3rud, 2026-09-21:
Ciao @Mancio-home, ho anche io una SY250 sei riuscito ad integrare il progetto?