leoshusar/4heat-esphome #8: Lots of  "Moretti Stove offline detected problem"

catdogmaus, 2024-04-05:
I managed to install my stove with D1 mini board but there are lots of "Moretti Stove offline detected problem" - about every 10 sec or so. Also I have in every 15-20 min in HA "Discovered new device" notification.
When I look at ESPHome logs I have regularly some errors and I dont know what is wrong.
`[23:34:09][W][fourheat:186]: Response timeout, retrying (0/3)
[23:34:10][W][fourheat:186]: Response timeout, retrying (1/3)
[23:34:10][W][fourheat:186]: Response timeout, retrying (2/3)
[23:34:10][W][fourheat:189]: Response timeout, giving up
[23:34:10][W][fourheat:192]: Clearing command TX queue, because the module can reappear in an unknown state`
I somehow think that all those errors must be related.

leoshusar, 2024-04-05:
What sensors do you have? Can you try with only one sensor and add them one by one?
This mostly happens when you send an unknown command (datapoint) - the stove stops responding altogether.

catdogmaus, 2024-04-05:
> What sensors do you have?

I have basically copy of your suggested basic yaml setup. I actually put it all here -hardware and yaml just in case if somebody other wants to add Moretti oven to HA
https://community.home-assistant.io/t/tiemme-4heat-stove-wifi-controller/374040/270
I have no idea where did you get all the datapoints, I actually like to add later adjustable threshold to buffer temperature but it probably is over my head.
I noticed that I have all sensor values except water pump state. should I try to remove it?

leoshusar, 2024-04-05:
My example YAML is mostly taken from [this](https://github.com/zaubererty/homeassistant-4heat/) integration.

My current production config at home is a combination of traffic sniffing from the official 4Heat WiFi module, my stove manual (you can fill out error codes from there, I noticed they differ between stoves) and some trial / error.

And yes, try to remove the pump. If this doesn't help, you can try to enable verbose logging and see what command is being send last before it stops responding.

catdogmaus, 2024-04-06:
> What do you mean by adjustable buffer temperature?

Buffer threshold. Every cheap thermostat has ability to change threshold where it turns on and off. For some reason Moretti has fixed it to 3 degrees, what is OK when you heat air directly, but completely insane when you heat buffer. That is making stove turning on and off in every half hour bcs temperature in buffer is never same. Difference in upper and lower layers is quite substantial. Every time circulation kicks on or somebody uses water, different layers get mixed and stove tries to start, only to turn off again after 15 minutes.

Removing pump state fixed the issues. I don't have official 4heat module so unfortunately it is no go for me to find correct datapoints.

closed 2024-04-10 by catdogmaus

leoshusar, 2024-04-11:
For your buffer threshold - when you get bored, you can check the [NG21 manual](https://www.fornello.ro/cs-content/cs-docs/2607-1634039243.pdf), I'm pretty sure it can be configured somewhere.
maxgu on the HA community forum seemed very knowledgeable about these settings, maybe he can help you if you get lost.

Glad to hear you've got at least something!

catdogmaus, 2024-04-11:
> you can check the [NG21 manual](https://www.fornello.ro/cs-content/cs-docs/2607-1634039243.pdf),

I think this is NG01, but there could be similarities, may be worth to look at.