leoshusar/4heat-esphome #9: mode switch

catdogmaus, 2024-07-29:
I hate to ask about the issue that might not be related to your code at all.
Is it possible to using yaml add the mode switch? I added lots of datapoints I needed but what I am trying to implement now is add night mode switch. What it should do is simply switch off cleaning engine and some other noisy stuff.
Basically what I have is 4-5 datapoints and turning the switch on should send all of them state 1 , turning switch off it should revert back to state 0.
Also it would be great to switch same time power from auto to some lower settings and afterwards back again. although the latter can probably be achieved somehow with the help of HA automations. However, this would add for users another layer of complexity, and doing it using yaml if possible would probably be more preferable.
Thanks

leoshusar, 2024-07-31:
Hi, no worries, feel free to ask.

For the first case I'd just do a [template switch](https://esphome.io/components/switch/template.html). In the `turn_on_action` you can set all needed switches to your desired state, in the `turn_off_action` set them to the opposite.
You just need to figure out some `lambda` logic which decides if it's currently night mode or not. I can think of two ways - check some (or all) switches if they are on / off and decide on that, or create a binary_sensor template which will hold the value. Or you can use `assumed_state`, but this will create a switch made of two buttons ON and OFF instead of a normal indicative switch in HA.

```yaml
select:
  - id: power
    name: Výkon
    platform: fourheat
    datapoint: B20364
    options:
      1: "P1"
      2: "P2"
      3: "P3"
      4: "P4"
      5: "P5"
      6: "P6"
      7: "Auto"

# Example configuration entry
switch:
  - platform: template
    name: Night mode
    lambda: |-
      if (your logic to check if it's night mode) {
        return true; // it is night mode, switch indicates ON
      } else {
        return false; // else OFF
      }
    turn_on_action:
      - switch.turn_off: switch1
      - switch.turn_off: switch2
      - switch.turn_off: switch3
      - select.set:
          id: power
          option: "P2"
    turn_off_action:
      - switch.turn_on: switch1
      - switch.turn_on: switch2
      - switch.turn_on: switch3
      - select.set:
          id: power
          option: "Auto"
```
(YAML not tested, quickly wrote off the top of my head)

I have also included your power setting. You can also create text sensor template `previous_power`, which will hold the previous power setting, if you want to use not only `Auto`, but for example `P4, P5, P6, Auto` and always go back to the previous value.

I hope I understood you well.

catdogmaus, 2024-07-31:
> For the first case I'd just do a [template switch](https://esphome.io/components/switch/template.html).

ESPHome is not really my topic. So I have first to create those 4 switches...
From Tiemme documentation I have datapoints that have default value 0. 
```
B20701  Cleaning Motor Off in Night Mode, 
B20700  Cleaning Motor 2 Off in Night Mode, 
B20699  Cleaning Motor 3 Off in Night Mode, 
B20698  Charging Motor Off in Night Mode
```
I am actually bit confused how to write yaml so that switch _on_ will send value 1 and switch _off_ will send 0.
I should probably use   `  on_turn_on:    on_turn_off: ` but what actual command it should be to send 1 and 0?

So something like this:
```
switch:
  - platform: fourheat
    id: ChargingMotorOffinNightMode
    name: Charging Motor Off in Night Mode
    internal
    on_turn_on:
    - # datapoint: B20698 some command to send 1
    on_turn_off:
    - # datapoint: B20698 some command to send 0
```

As I understand when I set those switches internal there will be no separate switches in frontend for the user to fiddle.
Ideally, as I imagine this, in front end there would only be a Nigth Mode switch, and HA automation/helper could then turn it on according to the time of day. 
I also imagine that this timer can be ultimately set using ESPHome, but it's definitely too much for me if I can't even get my simple switches in place.

leoshusar, 2024-08-01:
Can you send me the documentation? By the `Cleaning Motor Off in Night Mode` it seems to me like the Tiemme itself supports some kind of night mode, maybe it could be enabled directly without needing to mess with individual switches?

Anyway yeah, you create four switches like this:
```yaml
switch:
  - platform: fourheat
    id: charging_motor_off
    datapoint: B20698
    on_data: '1'
    off_data: '0'
    parser: return data[data.size() - 1] != '0';

  - platform: fourheat
    id: another_switch
    datapoint: B12345
    on_data: '1'
    off_data: '0'
    parser: return data[data.size() - 1] != '0';
```

The first switch: it sends an empty command to `A20698` and Tiemme should respond `B20698000000000001` if ON. This is how it *usually* works, by sending the data to A.
The `parser` part then checks the last character in the response - if it's not `'0'`, the switch is ON.
*If* the response is different, for example the `1` is not last, but third last like this `B20698000000000100`, you write the parser like this: `return data[data.size() - 3] != '0';`. Hope this makes sense.

If you don't specify `name`, it's considered an `internal` component and won't be visible to HA, it will only be used internally for your template switch.

And then you control them like this using their `id`:
```yaml
switch:
  - platform: template
    ...
    turn_on_action:
      - switch.turn_off: charging_motor_off
      - switch.turn_off: another_switch
    turn_off_action:
      - switch.turn_on: charging_motor_off
      - switch.turn_on: another_switch
```

catdogmaus, 2024-08-02:
Thanks, I got it to work.

> Can you send me the documentation? By the `Cleaning Motor Off in Night Mode` it seems to me like the Tiemme itself supports some kind of night mode, maybe it could be enabled directly without needing to mess with individual switches?

Tiemme does indeed support night mode, that's why these switches are there. Unfortunately, my fireplace doesn't have it in menu, but the NG21 board inside it supports this feature. 
There is the way to find (some) datapoints without sniffing - just with little help with google. Sometimes there are some pdf files in italian when you search google by your board model, with all(?) needed values - datapoint, min, max, default value etc. In my case I think I used this https://www.solarweb.net/forosolar/attachments/usuarios-biomasa/31348d1713977088-comunicacion-rs232-modulo-wifi-4heat-tiemme-otros-comandos.pdf
U can search inside it _Modalita notte_ or any other topic u need. First number by row is actually datapoint u want to use. U just need to add prefix so 698 becomes B20698 in yaml. Sometimes it needs to be tested what prefix is correct. As I understand there are mostly options btween B20, J30 and J40.
Unfortunately, I couldn't find any more documentation than those four, so maybe a separate switch exists, maybe not. As I said, my stove does not have this function in the menu.
Some time ago, I was convinced that if there is such a function in the menu, it must be activated somehow, so there must be a separate switch. Now I'm not so sure.
Maybe at some point, someone with this on the stove menu will be able to sniff out if there's anything else to turn on or off.
This is what I used eventually

````
  - platform: fourheat
    id: charging_motor_off
    datapoint: B20698
    on_data: '1'
    off_data: '0'
    parser: return data[data.size() - 1] != '0';

  - platform: fourheat
    id: Cleaning_Motor_Off
    datapoint: B20701
    on_data: '1'
    off_data: '0'
    parser: return data[data.size() - 1] != '0';

  - platform: fourheat
    id: Cleaning_Motor_2_Off
    datapoint: B20700
    on_data: '1'
    off_data: '0'
    parser: return data[data.size() - 1] != '0';

  - platform: fourheat
    id: Cleaning_Motor_3_Off
    datapoint: B20699
    on_data: '1'
    off_data: '0'
    parser: return data[data.size() - 1] != '0';

  - platform: template
    name: Night mode
    lambda: |-
      if (id(charging_motor_off).state) {
        return true; // it is night mode, switch indicates ON
      } else {
        return false; // else OFF
      }
    turn_on_action:
      - switch.turn_on: charging_motor_off
      - switch.turn_on: Cleaning_Motor_Off
      - switch.turn_on: Cleaning_Motor_2_Off
      - switch.turn_on: Cleaning_Motor_3_Off
      - select.set:
          id: power
          option: "P3"
    turn_off_action:
      - switch.turn_off: charging_motor_off
      - switch.turn_off: Cleaning_Motor_Off
      - switch.turn_off: Cleaning_Motor_2_Off
      - switch.turn_off: Cleaning_Motor_3_Off
      - select.set:
          id: power
          option: "Auto"
````
The real-life results were a bit mixed. I was actually hoping for a bigger effect. In fact, it doesn't seem to actually turn off anything, the fans just run a little quieter. Or maybe I still miss some additional switches that must be activated.

closed 2024-11-09 by leoshusar