leoshusar/4heat-esphome #10: error compiling esp32

ilgio, 2024-11-01:
`Compiling .pioenvs/4heat-controller/src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp.o
In file included from src/esphome/components/fourheat/helpers.h:3,
                 from src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp:2:
src/esphome/components/fourheat/common.h:8:37: error: 'optional' is not a member of 'std'
 using parser_t = std::function<std::optional<T>(const std::vector<uint8_t> &)>;
                                     ^~~~~~~~
src/esphome/components/fourheat/common.h:8:37: note: 'std::optional' is only available from C++17 onwards
src/esphome/components/fourheat/common.h:8:37: error: 'optional' is not a member of 'std'
src/esphome/components/fourheat/common.h:8:37: note: 'std::optional' is only available from C++17 onwards
src/esphome/components/fourheat/common.h:8:47: error: template argument 1 is invalid
 using parser_t = std::function<std::optional<T>(const std::vector<uint8_t> &)>;
                                               ^
In file included from src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp:2:
src/esphome/components/fourheat/helpers.h:18:86: error: 'optional' in namespace 'std' does not name a template type
 const std::string get_query_datapoint_id(const std::string &datapoint_id, const std::optional<std::string> &query_datapoint_id);
                                                                                      ^~~~~~~~
src/esphome/components/fourheat/helpers.h:18:81: note: 'std::optional' is only available from C++17 onwards
 const std::string get_query_datapoint_id(const std::string &datapoint_id, const std::optional<std::string> &query_datapoint_id);
                                                                                 ^~~
src/esphome/components/fourheat/helpers.h:18:94: error: expected ',' or '...' before '<' token
 const std::string get_query_datapoint_id(const std::string &datapoint_id, const std::optional<std::string> &query_datapoint_id);
                                                                                              ^
In file included from src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp:2:
src/esphome/components/fourheat/helpers.h:21:6: error: 'optional' in namespace 'std' does not name a template type
 std::optional<T> parse(const std::optional<parser_t<T>> &parser, const std::vector<uint8_t> &data);
      ^~~~~~~~
src/esphome/components/fourheat/helpers.h:21:1: note: 'std::optional' is only available from C++17 onwards
 std::optional<T> parse(const std::optional<parser_t<T>> &parser, const std::vector<uint8_t> &data);
 ^~~
Compiling .pioenvs/4heat-controller/src/esphome/components/fourheat/button/fourheat_button.cpp.o
In file included from src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp:3:
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.h:22:25: error: 'parser_t' does not name a type; did you mean 'parse_int'?
   void set_parser(const parser_t<bool> &parser);
                         ^~~~~~~~
                         parse_int
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.h:22:33: error: expected ',' or '...' before '<' token
   void set_parser(const parser_t<bool> &parser);
                                 ^
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.h:27:8: error: 'optional' in namespace 'std' does not name a template type
   std::optional<std::string> query_datapoint_id_;
        ^~~~~~~~
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.h:27:3: note: 'std::optional' is only available from C++17 onwards
   std::optional<std::string> query_datapoint_id_;
   ^~~
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.h:29:8: error: 'optional' in namespace 'std' does not name a template type
   std::optional<parser_t<bool>> parser_;
        ^~~~~~~~
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.h:29:3: note: 'std::optional' is only available from C++17 onwards
   std::optional<parser_t<bool>> parser_;
   ^~~
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp: In member function 'virtual void esphome::fourheat::FourHeatBinarySensor::setup()':
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp:11:69: error: 'class esphome::fourheat::FourHeatBinarySensor' has no member named 'query_datapoint_id_'; did you mean 'set_datapoint_id'?
   auto query_id = get_query_datapoint_id(this->datapoint_id_, this->query_datapoint_id_);
                                                                     ^~~~~~~~~~~~~~~~~~~
                                                                     set_datapoint_id
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp: In member function 'void esphome::fourheat::FourHeatBinarySensor::set_query_datapoint_id(std::__cxx11::string)':
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp:27:91: error: 'class esphome::fourheat::FourHeatBinarySensor' has no member named 'query_datapoint_id_'; did you mean 'set_datapoint_id'?
 void FourHeatBinarySensor::set_query_datapoint_id(const std::string datapoint_id) { this->query_datapoint_id_ = datapoint_id; }
                                                                                           ^~~~~~~~~~~~~~~~~~~
                                                                                           set_datapoint_id
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp: At global scope:
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp:29:45: error: 'parser_t' does not name a type; did you mean 'parse_int'?
 void FourHeatBinarySensor::set_parser(const parser_t<bool> &parser) { this->parser_ = parser; }
                                             ^~~~~~~~
                                             parse_int
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp:29:53: error: expected ',' or '...' before '<' token
 void FourHeatBinarySensor::set_parser(const parser_t<bool> &parser) { this->parser_ = parser; }
                                                     ^
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp: In member function 'void esphome::fourheat::FourHeatBinarySensor::set_parser(int)':
src/esphome/components/fourheat/binary_sensor/fourheat_binary_sensor.cpp:29:77: error: 'class esphome::fourheat::FourHeatBinarySensor' has no member named 'parser_'; did you mean 'parent_'?
[... 27 lines cut ...]
src/esphome/components/fourheat/common.h:8:37: note: 'std::optional' is only available from C++17 onwards
src/esphome/components/fourheat/common.h:8:37: error: 'optional' is not a member of 'std'
src/esphome/components/fourheat/common.h:8:37: note: 'std::optional' is only available from C++17 onwards
src/esphome/components/fourheat/common.h:8:47: error: template argument 1 is invalid
 using parser_t = std::function<std::optional<T>(const std::vector<uint8_t> &)>;
                                               ^
In file included from src/esphome/components/fourheat/button/fourheat_button.cpp:2:
src/esphome/components/fourheat/helpers.h:18:86: error: 'optional' in namespace 'std' does not name a template type
 const std::string get_query_datapoint_id(const std::string &datapoint_id, const std::optional<std::string> &query_datapoint_id);
                                                                                      ^~~~~~~~
src/esphome/components/fourheat/helpers.h:18:81: note: 'std::optional' is only available from C++17 onwards
 const std::string get_query_datapoint_id(const std::string &datapoint_id, const std::optional<std::string> &query_datapoint_id);
                                                                                 ^~~
src/esphome/components/fourheat/helpers.h:18:94: error: expected ',' or '...' before '<' token
 const std::string get_query_datapoint_id(const std::string &datapoint_id, const std::optional<std::string> &query_datapoint_id);
                                                                                              ^
In file included from src/esphome/components/fourheat/button/fourheat_button.cpp:2:
src/esphome/components/fourheat/helpers.h:21:6: error: 'optional' in namespace 'std' does not name a template type
 std::optional<T> parse(const std::optional<parser_t<T>> &parser, const std::vector<uint8_t> &data);
      ^~~~~~~~
src/esphome/components/fourheat/helpers.h:21:1: note: 'std::optional' is only available from C++17 onwards
 std::optional<T> parse(const std::optional<parser_t<T>> &parser, const std::vector<uint8_t> &data);
 ^~~
*** [.pioenvs/4heat-controller/src/esphome/components/fourheat/button/fourheat_button.cpp.o] Error 1
========================= [FAILED] Took 20.03 seconds =========================`

ilgio, 2024-11-04:
I added this:
`  platformio_options:
    build_flags: "-std=gnu++17"
    build_unflags: "-std=gnu++11"`
filled out correctly, but does not receive any sensor data
[https://pastebin.com/4hB665yp](url)

[image]

leoshusar, 2024-11-04:
Regarding the missing `optional`, hmm, looks like ESPHome still uses older C++. My current guess would be I'm using ESP-IDF framework and it maybe uses newer C++ so I'm not running into that issue, but don't really know now. Anyway yeah, the build flags should work fine.

For the not receiving any data, can you enable debug and see if there's any communication?
```yaml
logger:
  level: DEBUG
```

ilgio, 2024-11-04:
wowww… I inverted pin TX with RX
now I see some sensors…
[image]
I have to find the command to switch from wood to pellet

I need this option:
[image]
and some sensors don't work.. and above all what doesn't work is the pump which always remains OFF even when it is on

leoshusar, 2024-11-04:
> I inverted pin TX with RX

haha classic :) glad you've got something working!

> what doesn't work is the pump

It's very possible your board either doesn't report the state, or reports it on a different datapoint. I'm afraid I'm no help with this Tiemme stuff, I don't have any deeper knowledge about them. I only know when I send *something* I usually get *something* back.

Haven't checked the HA forum for a while, but if you haven't yet, I suggest you can try to ask maxgu there.

leoshusar, 2024-11-09:
I'm closing this issue, if you need anything in the future, feel free to create a new one.

closed 2024-11-09 by leoshusar