leoshusar/4heat-esphome #15: Yaml Configuration

ebagnoli, 2025-06-20:
Dear Sirs:

The configuration file here it is very good, but also very basic. For example I would like to change the Thermostat desired temperature and the room temperature, but it doesn't work.
I have an Ungaro Stove with SY400 card.

Is it possible tho have your Yaml File posted here?

Thanks

ebagnoli, 2025-06-24:
Good morning and thank you very much for your kind reply.

Let me try to explain myself better: I have a Fluenth Ungaro acqua24 pellet stove, which, like many Italian stoves, is fitted with a Tiemme with SY400 card.
I used your component installed on an esp32 card (congratulations on your great work).
I can communicate with the stove, turning it on and off, and I receive the stove status correctly (on, off, etc.).

This part of your code is unusable for me because I don't have a buffer.

- id: puffer_temperature
    name: Puffer temperature
    platform: fourheat
    datapoint: J30012
    device_class: temperature
    unit_of_measurement: °C

- id: boiler_temperature
    name: Boiler temperature
    platform: fourheat
    datapoint: J30017
    device_class: temperature
    unit_of_measurement: °C

 number:
  - id: boiler_target_temperature
    name: Boiler target temperature
    platform: fourheat
    datapoint: B20199
    device_class: temperature
unit_of_measurement: °C
min_value: 0
max_value: 90

Instead, I need to be able to remotely set the value of the boiler thermostat and the room thermostat.

Can anyone tell me the datapoints for these two parameters?

Thank you

Enrico

[image]

leoshusar, 2025-06-24:
Hi, thank you for you kind words!

I see now what you mean.
Unfortunately the datapoints are not always the same, so even if a datapoint works for me, it might not work for you.

Anyway I've got most of mine from the original HA integration [here](https://github.com/zaubererty/homeassistant-4heat/blob/master/custom_components/4heat/const.py). I see there is a `Room temperature set point` and `Boiler target` (twice). You can try these. Let me know if they work for you.

ebagnoli, 2025-06-24:
Hi thank you for your prompt reply: the most important value to change for me is Boiler Tempreature, 
now I'm a little confused because i cannot find the difference between these two entry points.

    "20180": ["Boiler target", TEMP_CELSIUS, ""],
    "20199": ["Boiler target", TEMP_CELSIUS, ""],

The only thing to do when I will be in front of the Stove is to try these values until one of them works.

I'll keep you posted of the evolution.

Kind regards Enrico