leoshusar/4heat-esphome #17: update esphome doesent work

ilgio, 2025-09-17:
After updating esphome from 2025.5.0 to 2025.9.0 it no longer works and I can no longer see the sensors.

HELPME

leoshusar, 2025-09-17:
Hi, anything interesting in logs?

leoshusar, 2025-09-17:
Nevermind, it's fine. Last thing before I'll get to it, can you please try one more time but with logger level verbose?

leoshusar, 2025-10-01:
Hi, sorry, finally got to it.
I went straight to latest 2025.9.3 and it works fine on my end.

Btw I recommend avoiding .0 updates and wait for the latest one of the month, same for HA updates.

leoshusar, 2025-10-03:
Hmm, that's weird. Can you try with UART debug logging too? Just add `debug` key, like:

```yaml
uart:
  ...
  debug:
```

leoshusar, 2025-10-06:
Can you try to remove this:
```yaml
  platformio_options:
    build_flags: "-std=gnu++17"
    build_unflags: "-std=gnu++11"
```

ilgio, 2025-10-06:
nothing
:_(

[4heat.yaml](https://github.com/user-attachments/files/22723150/4heat.yaml)

[logs_4heat_logs.txt](https://github.com/user-attachments/files/22723153/logs_4heat_logs.txt)

ilgio, 2025-10-07:
give me the yaml with basic code with few sensors for testing, thanks

ilgio, 2025-10-07:
but 16 and 17 are not the UART

[image]

ilgio, 2025-10-08:
[image]
I have this converter between RS232 and ESP32S

leoshusar, 2025-10-08:
This one should be fine

ilgio, 2025-10-18:
I changed the uart with pins 16 and 17. and it works
🥳